mcuee/libusb-win32 #4: API Change: Redefining uint32_t et al. clashes with <stdint.h>

mcuee, 2018-06-23:
https://sourceforge.net/p/libusb-win32/bugs/77/

Updated: 2012-07-06 Created: 2012-01-19 Creator: Joerg Wunsch

When trying to compile AVRDUDE against libusb-win32-1.2.5.0, I get the following error:

In file included from usbasp.c:56:0:
/tmp/libusb-win32-src-1.2.5.0/src/usbi.h:10:23: error: conflicting types for 'uint32_t'
/usr/local/lib/gcc/mingw32/4.5.0/../../../../mingw32/include/stdint.h:32:20: note: previous declaration of 'uint32_t' was here
*** Error code 1

libusb should not attempt to redeclare type names found in <stdint.h>.

As a short-term workaround, I #if 0'ed these three lines out from usbi.h. I think a proper solution would be either to name your own types differently, or (better) to probe for C99 types availability
from <stdint.h> in some kind of configure script, and use them if available, falling back to the current way if no stdint.h is around.

(Btw., with those three lines disabled, all works well, and I could successfully build AVRDUDE on my FreeBSD host against the library. The resulting executable file works, which is great!)

mcuee, 2018-06-23:
libusb-win32 project is in a support only mode. Mailing list support is still provided, no changes to the codes and release.

mcuee, 2021-11-16:
I do not think we want API change now. So I will keep it as "wontfix".

dontech, 2023-10-01:
This is actually fixed recently, as the conflicting type really is benign and does not affect the ABI.

The problem was that type was "unsigned long" instead of "unsigned int". But on windows, this makes no difference, as "long" is always 32 bit and so is "int", so fixing this breaks nothing.

Fixed.

closed 2023-10-01 by dontech