mcuee/libusb-win32 #17: i met a problem in some device: WDI_ERROR_NEEDS_ADMIN

ShiverZm, 2020-09-24:
i used Administrator privilige to run this inf-wizard.exe,but it stil exited. And my roler also is Administrator.

mcuee, 2020-09-28:
Please take not libusb-win32 inf-wizard.exe has been deprecated. Please use libusbk inf-wizard or Zadig.

closed 2020-09-28 by mcuee

ShiverZm, 2020-09-28:
i tried libusbk  but libusbk inf-wizard is not universal ? must pair with libusbk 's Library ? mean that i have to modify what i use liusb-win32 's Library,i used to use libusbwin32 as the driver to communicate with my device.

mcuee, 2020-09-28:
Not so sure what you mean by universal. You can use libusbk inf wizard to install libusb-win32 driver.

Zadig is even better and more up to date.
https://zadig.akeo.ie/

mcuee, 2020-09-28:
Please also read the wiki which clearly mentioned this issue in the very beginning.
https://github.com/mcuee/libusb-win32/wiki