mcuee/libusb-win32 #24: Info: libusb0.sys snapshot release is not signed

mcuee, 2021-10-19:
Please take not libusb0.sys snapshot release after 1.0.26 release is not signed. Users who want to use it will have to get it signed by themselves.

Ref: the following snapshot release is not signed.
https://sourceforge.net/projects/libusb-win32/files/libusb-win32-snapshots/20190918/

mcuee, 2021-10-20:
Reference: low Windows driver rank score for libusbk as it is signed in Nov 2014. libusb0.sys 1.0.26 version was signed even earlier in Jan 2012.
https://github.com/pbatard/libwdi/wiki/Zadig#Preventing_the_replacement_of_a_Zadig_libusbK_USB_Device_Driver
> For instance, Zadig's date stamp for libusbK might be 11/30/2014 for version 3.0.7.0 resulting in rank 00FF0001.

mcuee, 2021-10-30:
Still there is an easy workaround for Windows 7 as well if Zadig snapshot does not work for Windows 7.

1) Using Zadig 2.6 release to install libusb0.sys 1.2.6.0 device driver.
2) Using the libusb-win32 1.2.7.2 installer to upgrade libusb0.sys and libusb0.dll

mcuee, 2021-10-30:
@dontech 
@tormodvolden has a comment to see whether it is possible that you produce a signed release of the libusb 1.2.7.2 snapshot debug version with the debug info ON. I am not so sure about the Microsoft portal attestation signing policy with regard to debug version.

dontech, 2021-10-31:
Hello all,

1) Yes, the attestation signing only works for win10. If we want full signing, we have to pass all the WHQL tests, which requires a larger test setup. I have done this before for clients, but it is super heavy. We could also find some bugs doing this, so it would not be a total waste.

2) Yeah i think we will include debug builds for the next release. No totally sure why "make dist" does not do this currently.

mcuee, 2021-11-02:
> Hello all,
> 
> 1. Yes, the attestation signing only works for win10. If we want full signing, we have to pass all the WHQL tests, which requires a larger test setup. I have done this before for clients, but it is super heavy. We could also find some bugs doing this, so it would not be a total waste.

@dontech 
If this can be done, that will be great. Take your time.

I assume that it will work on Windows 7 onwards, including Windows 8/8.1 and Windows 10, as well as the corresponding Windows server versions. 

Not so sure if we should even care for Windows XP and Windows Vista.

> 2. Yeah i think we will include debug builds for the next release. No totally sure why "make dist" does not do this currently.

That will be great.

mcuee, 2021-11-02:
BTW, the following libusb issue fix does not seems to need the fix in 1.2.7.2.
- https://github.com/libusb/libusb/issues/94

mcuee, 2021-11-10:
New Zadig 2.7 build with libusb0.sys 1.2.7.2 snapshot release.
https://github.com/mcuee/libwdi/releases/tag/v1.4.1

I built Zadig 2.7 from libwdi 1.4.1 release source codes with VS2019 and tested it with Windows 10 and 11. 
I am not so sure if it works under Windows 7/8/8.1. It will not work under Windows XP. Unlikely it will work under Windows Vista either. Please report your success or failure. Thanks.

changyp6, 2021-11-11:
> New Zadig 2.7 build with libusb0.sys 1.2.7.2 snapshot release. https://github.com/mcuee/libwdi/releases/tag/v1.4.1
>
> I built Zadig 2.7 from libwdi 1.4.1 release source codes with VS2019 and tested it [...]

I have tried this driver on Windows 7 SP1 (32 / 64) / Windows 8.1 Version 6.3 Build 9600 x86_64 / Windows 10

On both windows 8.1 and windows 10, libusb0.sys is reported "signed by libusb-win32"
On windows7, libusb0.sys is reported "not signed", in x86 Windows7, driver can be loaded and can work, however, in x86_64 Windows7, driver is refused to be loaded by system, and "Error Code 52" is reported in the device manager driver detail page.

libusb0.sys 1.2.7.2 still CANNOT work in Windows 7

mcuee, 2021-11-11:
Thanks a lot for the report. This is kind of expected based on the report in libusb git issues discussions.

dontech, 2021-11-13:
I have found a work-around:

https://community.osr.com/discussion/293107/now-that-cross-signing-is-deprecated-how-are-you-all-supporting-windows-7

The reason for this mess is the deprecation of cross-certs, and a specific bug in the windows 7 cert handling.

I will try to create a new snapshot with the certificate chain changes need, and see if it helps...

dontech, 2021-11-13:
OK new snapshot ready:

https://sourceforge.net/projects/libusb-win32/files/libusb-win32-snapshots/

1) changyp6, could you please try this and report back?

2) Additionally, it would be great if you also tried it on an older win7 install, to see if the SHA1 i added actually works. Not sure.

changyp6, 2021-11-13:
> OK new snapshot ready:
>
> https://sourceforge.net/projects/libusb-win32/files/libusb-win32-snapshots/
>
>     1. changyp6, could you please try this and report back?
>
>     2. Additionally, it would be [...]

I don't have test environment right now, so I'll try this new snapshot on next Monday, and give you feedback ASAP.

mcuee, 2021-11-13:
@pazourek Please help to test 1.2.7.3 snapshot release under Windows 7 as well if you got the time. Thanks.

pazourek, 2021-11-13:
I'm sorry, but I'm a little confused how to test the 1.2.7.3 snapshot. The version 1.2.7.2 contained an inf-wizard so I was able to install the driver. But the latest snapshot (libusb-win32-bin-1.2.7.3.zip) doesn't. There is also missing a .cat file.

mcuee, 2021-11-14:
> I'm sorry, but I'm a little confused how to test the 1.2.7.3 snapshot. The version 1.2.7.2 contained an inf-wizard so I was able to install the driver. But the latest snapshot (libusb-win32-bin-1.2.7.3.zip) doesn't. There is also missing a .cat file.

You can use Zadig to install libusb0.sys 1.2.6.0 driver for your device first, and then install libusb-win32 1.2.7.2 to upgrade the libusb0.sys and libusb0.dll file.

But I will create a release of Zadig 2.7 with the 1.2.7.3 snapshor release as well in a few hours.

mcuee, 2021-11-14:
@pazourek 
You can try my build of libwdi/Zadig here.
https://github.com/mcuee/libwdi/releases/tag/v1.4.1

mcuee, 2021-11-14:
Mirror of libusb-win32 1.2.7.3 snapshot release here, I also include the above Zadig release.
https://github.com/mcuee/libusb-win32/releases/tag/snapshot_1.2.7.3

pazourek, 2021-11-15:
The dirver installed by Zadig sems to work properly under Win7 Pro 64bit.

mcuee, 2021-11-15:
> The dirver installed by Zadig sems to work properly under Win7 Pro 64bit.

Great. Thanks for the confirmation.

changyp6, 2021-11-15:
> Mirror of libusb-win32 1.2.7.3 snapshot release here, I also include the above Zadig release. https://github.com/mcuee/libusb-win32/releases/tag/snapshot_1.2.7.3

I have tested libusb0.sys 1.2.7.3 driver on Windows 7 SP1 (32 / 64) / Windows 8.1 Version 6.3 Build 9600 x86_64 / Windows 10 x86_64
libusb0.sys 1.2.7.3 can be loaded on both Windows 7 (32 / 64) and can work well, however, in the driver detailed info page, it still shows that "driver is not signed".
libusb0.sys 1.2.7.3 can be loaded on Windows 8.1 / Windows 10 x86_64, and can work well, in the driver detailed info page, it shows "driver is signed by libusb-win32"

dontech, 2021-11-15:
>>libusb0.sys 1.2.7.3 can be loaded on both Windows 7 (32 / 64) and can work well, however, in the driver detailed info page, it >>still shows that "driver is not signed".

Yeah, i noticed the same. Any ideas?

It think it's because its attestation signed, which for some reason is not liked by the older Windows GUI.
Anyone know how to fix this?

Clearly the GUI is not driven by the same logic as the driver installer itself, at the installer approves of the driver now.
So i would say this went from a blocker to a minor.

dontech, 2021-11-15:
It does the same for libusbk installed via Zadig.

I think this is something related to some Windows internals. Not sure this is fixable.
Also, it's purely cosmetic.

Maybe someone at libusbK knows more?

dontech, 2021-11-15:
I think its because the GUI wants the driver to be cross signed by Microsoft.

Cross signing was removed and deprecated: https://docs.microsoft.com/en-us/windows-hardware/drivers/install/deprecation-of-software-publisher-certificates-and-commercial-release-certificates

As long as the driver works, i think its OK. Ideas?

mcuee, 2021-11-15:
> I think its because the GUI wants the driver to be cross signed by Microsoft.
>
> Cross signing was removed and deprecated: [...]

Yes, I agree with your conclusion. We are more or less ready with the next release.

mcuee, 2021-11-15:
For those who want to get the 'cosmetic' fixed, one way is to submit your driver package (for specific devices specified in the inf) for WHQL and hopefully it can pass the tests. You need to have the EV cert and you need to pass the HLK.

mcuee, 2021-11-15:
@dontech -- can you do another experiment with attestation signing? This time you do not use your cert to sign the package. It is said that the result package will be working under Windows 7 and Windows 10.

Ref:
https://community.osr.com/discussion/292832/cross-signing-certs-expiring-this-week-how-will-whql-work
https://www.osr.com/blog/2021/04/08/lost-cause-no-driver-updates-allowed-except-for-win-10/

*********************************************************
Peter_Viscarola_(OSR) Administrator
Posts: 8,747
April 14
C’mon... You’re making this a bigger deal than it is.

You have one or more cert that are registered with the dashboard. Sign everything with one of those certs, including your driver package, when you submit it for Attestation Signing. No cross-signing. The Attestation Signed package will install only on Win 10.

OR

Don’t sign the drivers, and the package will be installable on Win 7 and Win 10.

Done. No need to ask MSFT anything,

Peter

Peter Viscarola
OSR
@OSRDrivers

dontech, 2021-11-17:
I tried not signing anything. Same thing: loads on win7 x64, but still states "not signed in GUI". 

Fairly sure this is an O/S limitation we cannot get around, which was caused by the expiration of the cross-certs, which win7 GUI apparently wants. Either that or it wants an MS SHA1 signature, which the signing portal does not provide. To fix that Microsoft would need to push a fix for WIN7. Since WIN7 is EOL thats doubtful.

I think we a stuck with the current solution, unless someone can point to someone who has gotten around this.

mcuee, 2021-11-17:
Good. Thanks for the confirmation. I will close this issue now.

closed 2021-11-17 by mcuee