mcuee/libusb-win32 #30: Port to later SDK.

dontech, 2021-11-17:
Current SDK supported is WIN7 SDK, which is getting a bit old.

mcuee, 2021-11-17:
Yes this is a good idea. I think we need to drop Windows XP/Vista support and then set the minimum OS requirement to Windows 7.

In that case, we can use the latest Windows 11 WDK.
https://docs.microsoft.com/en-us/windows-hardware/drivers/download-the-wdk

mcuee, 2021-11-17:
@dontech If you want to keep the build script, Enterprise WDK (EWDK) may be considered.
https://docs.microsoft.com/en-us/windows-hardware/drivers/download-the-wdk

dontech, 2021-11-17:
mcuee, you read my mind. Already looking at switching to win11 DDK, and dropping xp/vista support.

mcuee, 2023-10-01:
This can be closed now that #40 and #41 have been merged.

closed 2023-10-01 by mcuee