mcuee/libusb-win32 #33: Enhancement: to add WinUSB support

mcuee, 2022-05-14:
There is a fork version of libusb-win32 here which adds WinUSB support for a subset of the libusb-win32 API.
https://github.com/avrdudes/libusb/tree/windows/src

Motivation of the fork.
https://github.com/mariusgreuel/avrdude#support-winusb-devices-via-custom-libusb

Not so sure if it makes sense to add WinUSB support for libusb-win32 now (libusb-compat-0.1 under Windows will support WinUSB).

petabyt, 2022-09-18:
I tried to use that library on a PTP/USB device and it wouldn't work. `WinUsb_Initialize` woudn't work on my device.  

Quick question, is there a way to use libusb without using any installers or INF files? Just have it search for a specific GUID (in my case, `GUID_DEVCLASS_IMAGE`), and work? For a while I've been telling users of https://github.com/petabyt/mlinstall to use Zadig, but it's a pain to use and uninstall. Just test your devel-filter installer and it works great, but wish there was a quicker procedure.

mcuee, 2022-09-19:
> Quick question, is there a way to use libusb win32 without using any installers or INF files?

Sorry but not possible.

dontech, 2024-02-20:
I am really not sure this brings value. Allot of people are using libusb0 via libusb-1.0 which already supports winusb. That people would use winusb via the legacy libusb-0.1 is a bit unrealistic.

mcuee, 2024-02-20:
I agree. I will close this issue as not planned.

closed 2024-02-20 by mcuee