mcuee/libusb-win32 #39: github action

mcuee, 2023-09-24:
It may be good to set up github action for CI/CD.

dontech, 2023-10-01:
> It may be good to set up github action for CI/CD.

Yes, this would be a logical step

tormodvolden, 2023-10-02:
Running the static driver verifier can maybe be included?
https://learn.microsoft.com/en-us/windows-hardware/drivers/devtest/static-driver-verifier

dontech, 2023-10-02:
Yes, any contributions are welcome

dontech, 2023-10-02:
> Running the static driver verifier can maybe be included? https://learn.microsoft.com/en-us/windows-hardware/drivers/devtest/static-driver-verifier

This should probably be added to the build system

dontech, 2023-10-02:
We can follow this: https://learn.microsoft.com/en-us/windows-hardware/drivers/devtest/using-static-driver-verifier-to-find-defects-in-drivers

dontech, 2023-10-19:
The static driver verifier will be added to the build system after the first release of 3.0.0.X

Also, all errors and warnings will be fixed that is emitted by this system (currently its ~50 errors emitted - mostly minor)

tormodvolden, 2023-10-19:
I hope you mean 1.3.0.x :) BTW, libusbK.dll uses the driver version < 3.0.0.0 to distinguish it from libusbK.sys

mcuee, 2023-10-20:
> I hope you mean 1.3.0.x :) 
Haha.

> BTW, libusbK.dll uses the driver version < 3.0.0.0 to distinguish it from libusbK.sys

Interesting info. Thanks.

mcuee, 2023-10-20:
In the future, maybe github CodeQL can be added as well.
https://learn.microsoft.com/en-us/windows-hardware/drivers/devtest/static-tools-and-codeql

mcuee, 2024-02-17:
libwdi github CI example: with codeql, coverity and VS2022 as well as MSYS build.
https://github.com/pbatard/libwdi/tree/master/.github/workflows