mcuee/libusb-win32 #42: Make GCC/MinGW builds works on Linux

dontech, 2023-10-01:
@tormodvolden mentioned this: https://github.com/mcuee/libusb-win32/pull/36

Test scenario: build tools and driver on Linux (debian or fedora based), and make the output work in win10/win11

dontech, 2023-10-01:
Inspiration from here: https://github.com/utoni/mingw-w64-dpp

dontech, 2023-10-01:
@tormodvolden what recipe or command line options do you use to test build on debian?

tormodvolden, 2023-10-01:
`make host_prefix=x86_64-w64-mingw32 ddk=/usr/x86_64-w64-mingw32 driver`

tormodvolden, 2023-10-01:
Just for reference, these are the build flags that get applied when building mingw-w64-dpp. I haven't checked that the dpp examples actually load on Windows though.

make -C examples DPP_ROOT=/home/tormod/SRC/mingw-w64-dpp all
make[1]: Entemake -C examples DPP_ROOT=/home/tormod/SRC/mingw-w64-dpp all
make[1]: Entering directory '/home/tormod/SRC/mingw-w64-dpp/examples'
/usr/bin/x86_64-wring directory '/home/tormod/SRC/mingw-w64-dpp/examples'
/usr/bin/x86_64-w64-mingw32-gcc -std=c99 -Wall -Wextra -Wno-sign-compare -Wno-strict-aliasing -m64 -fPIC -fvisibility=hidden -ffunction-sections -fdata-sections -fno-builtin -ffreestanding -fno-stack-protector -mno-stack-arg-probe -I/home/tormod/SRC/mingw-w64-dpp/CRT -I/usr/x86_64-w64-mingw32/include/ddk -D__INTRINSIC_DEFINED_InterlockedBitTestAndSet -D__INTRINSIC_DEFINED_InterlockedBitTestAndReset   -c dpp-example.c -o dpp-example.o
CC  dpp-example.o
/usr/bin/x86_64-w64-mingw32-gcc -Wall -Wextra -Wno-sign-compare -Wno-strict-aliasing -m64 -fPIC -fvisibility=hidden -ffunction-sections -fdata-sections -fno-builtin -ffreestanding -fno-stack-protector -mno-stack-arg-probe -I/home/tormod/SRC/mingw-w64-dpp/CRT -I/usr/x86_64-w64-mingw32/include/ddk -D__INTRINSIC_DEFINED_InterlockedBitTestAndSet -D__INTRINSIC_DEFINED_InterlockedBitTestAndReset -shared -Wl,--subsystem,native -Wl,--image-base,0x140000000 -Wl,--dynamicbase -Wl,--nxcompat -Wl,--file-alignment,0x200 -Wl,--section-alignment,0x1000 -Wl,--stack,0x100000 -Wl,--gc-sections -Wl,--exclude-all-symbols -Wl,--entry,_CRT_DriverEntry -nostartfiles -nodefaultlibs -nostdlib -Wl,-Map='dpp-example.sys.map' -o 'dpp-example.sys' dpp-example.o /home/tormod/SRC/mingw-w64-dpp/CRT-build/libcrt.a -lntoskrnl -lhal 
LD  dpp-example.sys

dontech, 2024-01-07:
The system event viewer says this in "application popup" message:

libusb0.sys cannot find usbd.sys _USBD_CreateConfigurationRequestEx

Notice the underscore. Removing "--add-underscore" fixes the problem.

FIXED.

dontech, 2024-01-07:
https://github.com/mcuee/libusb-win32/commit/04dde53fd4392d1513690a81df12011adeb7884e

dontech, 2024-01-07:
@tormodvolden please verify

closed 2024-01-07 by dontech

tormodvolden, 2024-02-12:
> @tormodvolden please verify

Yes! I can confirm this works! Great!

And thanks for the System Event Viewer clue. I hadn't looked in there (and I don't know if I would take notice of that extra underscore...).

Now just sad that the usbdex thing complicates it all...