mcuee/libusb-win32 #43: libusb0.dll missing

zamirfthi, 2023-10-03:
I am building an application that utilizes libusb-0.1. It was able to be compiled and built on Microsoft Visual Studio 2022 up until yesterday where system error was raised upon debugging. The description of the error is "The code execution cannot proceed because libusb0.dll was not found. Reinstalling the program may fix this problem." I did try debugging the program on another machine but the problem still persists. Was ther
[image]
e any commit changes to the repository yesterday that might cause this to occur?

mcuee, 2023-10-03:
No issues here with VS2019. I will try VS2022 later.

Maybe you want to clone the git repo again and see if that helps. VS project files have been upgraded to VS2019 now.
https://github.com/mcuee/libusb-win32/tree/master/libusb/projects/vs2019

mcuee, 2023-10-03:
No issues with an up-to-date version of VS2022 either. Here is the build log (take note libusb0.sys build is not supported under VS2022, only with VS2019 or EWDK.

Build log:
```
Rebuild started...
1>------ Rebuild All started: Project: install-filter-win, Configuration: Debug x64 ------
2>------ Rebuild All started: Project: install-filter, Configuration: Debug x64 ------
3>------ Rebuild All started: Project: libusb-dll, Configuration: Debug x64 ------
4>------ Rebuild All started: Project: libusb0, Configuration: Debug x64 ------
4>C:\Program Files (x86)\Windows Kits\10\build\10.0.22621.0\WindowsDriver.common.targets(238,5): error :  'Windows7' is not a supported OS Version
4>Done building project "libusb0.vcxproj" -- FAILED.
2>error.c
1>error.c
3>descriptors.c
3>error.c
1>install.c
2>install.c
3>install.c
1>install_filter_win.c
2>install_filter.c
3>registry.c
2>registry.c
1>registry.c
2>Generating Code...
3>usb.c
1>Generating Code...
3>windows.c
2>install-filter.vcxproj -> C:\work\libusb\libusb-win32\libusb-win32\libusb\projects\vs2019\x64\Debug\install-filter.exe
1>install-filter-win.vcxproj -> C:\work\libusb\libusb-win32\libusb-win32\libusb\projects\vs2019\x64\Debug\install-filter-win.exe
3>Generating Code...
3>   Creating library C:\work\libusb\libusb-win32\libusb-win32\libusb\projects\vs2019\x64\Debug\libusb0.lib and object C:\work\libusb\libusb-win32\libusb-win32\libusb\projects\vs2019\x64\Debug\libusb0.exp
3>libusb-dll.vcxproj -> C:\work\libusb\libusb-win32\libusb-win32\libusb\projects\vs2019\x64\Debug\libusb0.dll
5>------ Rebuild All started: Project: testlibusb-win, Configuration: Debug x64 ------
6>------ Rebuild All started: Project: testlibusb, Configuration: Debug x64 ------
6>testlibusb.c
5>testlibusb_win.c
6>testlibusb.vcxproj -> C:\work\libusb\libusb-win32\libusb-win32\libusb\projects\vs2019\x64\Debug\testlibusb.exe
5>testlibusb-win.vcxproj -> C:\work\libusb\libusb-win32\libusb-win32\libusb\projects\vs2019\x64\Debug\testlibusb-win.exe
========== Rebuild All: 5 succeeded, 1 failed, 0 skipped ==========
========== Rebuild started at 12:39 pm and took 04.638 seconds ==========
```

Debug run for testlibusb.
```
LOG_APPNAME define missing:[os_init] dll version: 1.2.7.4
LOG_APPNAME define missing:[os_init] driver version: 1.2.6.0
LOG_APPNAME define missing:[set_debug] setting debugging level to 255 (on)
LOG_APPNAME define missing:[os_find_busses] found bus-0
LOG_APPNAME define missing:[os_find_devices] found \\.\libusb0-0001--0x16c0-0x05dc on bus-0
LOG_APPNAME define missing:err [os_open] invalid file name virtual-hub

Dev #0: 0000 - 0000

  Dev #1: www.fischl.de - USBasp

C:\work\libusb\libusb-win32\libusb-win32\libusb\projects\vs2019\x64\Debug\testlibusb.exe (process 11416) exited with code 0.
To automatically close the console when debugging stops, enable Tools->Options->Debugging->Automatically close the console when debugging stops.
Press any key to close this window . . .
```
Debug run output of the GUI testlibusb-win:
```
DLL version:	1.2.7.4
Driver version:	1.2.6.0

bus/device  idVendor/idProduct
bus-0/\\.\libusb0-0001--0x16c0-0x05dc     16C0/05DC
- Manufacturer : www.fischl.de
- Product      : USBasp
- Serial Number: 0001
bLength:             18
bDescriptorType:     01h
bcdUSB:              0201h
bDeviceClass:        00h
bDeviceSubClass:     00h
bDeviceProtocol:     00h
bMaxPacketSize0:     08h
idVendor:            16C0h
idProduct:           05DCh
bcdDevice:           0110h
iManufacturer:       1
iProduct:            2
iSerialNumber:       3
bNumConfigurations:  1
  wTotalLength:         18
  bNumInterfaces:       1
  bConfigurationValue:  1
  iConfiguration:       0
  bmAttributes:         80h
  MaxPower:             250
    bInterfaceNumber:   0
    bAlternateSetting:  0
    bNumEndpoints:      0
    bInterfaceClass:    255
    bInterfaceSubClass: 0
    bInterfaceProtocol: 0
    iInterface:         2
```

zamirfthi, 2023-10-04:
The cause of the issue has been found. Our machine was installed with the latest Windows Update which somehow caused driver issues. Simply updating the driver helped us in overcoming this issue and everything is resolved now. Thank you for your assist. Sorry for any inconvenience caused.

closed 2023-10-04 by zamirfthi