mcuee/libusb-win32 #44: Hotplug support

mcuee, 2023-10-10:
It may be good to provide hotplug support.

As of now, libusb-win32 does not support hotplug. libusb-1.0 does not support hotplug under Windows. 
libusbK has hotplug support.

dontech, 2023-10-12:
Please clarify what exactly you mean by hotplug.

Do you mean generating a callback when a new device is enumerated ?

mcuee, 2023-10-12:
> Please clarify what exactly you mean by hotplug.
> 
> Do you mean generating a callback when a new device is enumerated ?

Yes, something along that line -- notification about device hotplug event.

libusbK hotplug APIs:
https://libusbk.sourceforge.net/UsbK3/group__hotk.html

libusb-1.0 hotplug APIs: (not yet implemented on Windows platform)
https://libusb.sourceforge.io/api-1.0/group__libusb__hotplug.html

sonatique, 2023-12-08:
@mcuee : Quick question: what about fully implementing hotplug directly inside libusb-1.0 (supposing it is possible using win32-like API)?
Would there be any downside as compare to implementing the core functionality inside libusb0.sys and only calling it from libusb-1.0?

mcuee, 2023-12-08:
I think typical hotplug implementation is in the user mode side and not on the kernel side. For example you can look at libusbK.

BTW, libusb-win32 API (libusb0.dll) is a superset of libusb-0.1, so it is not compatible with libusb-1.0 API.

mcuee, 2024-02-20:
@dontech 

Please check if this is worth the efforts or not, if not, we can close this issue as not planned.

dontech, 2024-02-20:
We can still provide hotplug support to libusb-1.0 in the future I think, so lets keep this one open for now.