mcuee/libusb-win32 #46: Update documentation on building

dontech, 2023-10-29:
As pointed out, the current documentation is either wrong or lacking.

At least detail how to:

1) Build using Visual Studio
2) Build using command line "build.cmd"

mcuee, 2024-02-19:
@dontech 

Do you want to create a new `BUILD.md` file in the repo? Or we can just update the Wiki to replace the outdated info with the latest info?
https://github.com/mcuee/libusb-win32/wiki#user-content-Development

dontech, 2024-02-19:
Well we probably need to update the wiki anyway, or remove it.

I am kind of old-school so I was thinking of having the documentation inside the repo, so that the documentation lives with the source no matter where it is.

To keep it simple can't we just have a readme.md to put everything that was in wiki, and then update that to match current setup ?

mcuee, 2024-02-19:
> Well we probably need to update the wiki anyway, or remove it.
>
> I am kind of old-school so I was thinking of having the documentation inside the repo, so that the documentation lives with the source [...]

Good idea.

I will suggestion that keep the following three section in the Wiki and then move the rest in `README.md`.

[Open Source projects using libusb](https://github.com/libusb/libusb/wiki#user-content-Open_Source_projects_using_libusb)
[Language Bindings](https://github.com/libusb/libusb/wiki#user-content-Language_Bindings)
[Other Resources](https://github.com/libusb/libusb/wiki#user-content-Other_Resources)