mcuee/libusb-win32 #58: Building issue under MSYS2 mingw64 gcc

mcuee, 2024-02-09:
There is a minor here here.

```
 UCRT64 /c/work/libusb/libusb-win32/libusb
$ make
gcc -O2 -Wall -DWINVER=0x500  -Werror -Wno-unknown-pragmas -Wno-multichar -DLOG_APPNAME=\"libusb0-dll\" -DTARGETTYPE=DYNLINK -c ./src/usb.c -o usb.2.o  -I./src -I./src/driver -I.
gcc -O2 -Wall -DWINVER=0x500  -Werror -Wno-unknown-pragmas -Wno-multichar -DLOG_APPNAME=\"libusb0-dll\" -DTARGETTYPE=DYNLINK -c ./src/error.c -o error.2.o  -I./src -I./src/driver -I.
gcc -O2 -Wall -DWINVER=0x500  -Werror -Wno-unknown-pragmas -Wno-multichar -DLOG_APPNAME=\"libusb0-dll\" -DTARGETTYPE=DYNLINK -c ./src/descriptors.c -o descriptors.2.o  -I./src -I./src/driver -I.
gcc -O2 -Wall -DWINVER=0x500  -Werror -Wno-unknown-pragmas -Wno-multichar -DLOG_APPNAME=\"libusb0-dll\" -DTARGETTYPE=DYNLINK -c ./src/windows.c -o windows.2.o  -I./src -I./src/driver -I.
gcc -O2 -Wall -DWINVER=0x500  -Werror -Wno-unknown-pragmas -Wno-multichar -DLOG_APPNAME=\"libusb0-dll\" -DTARGETTYPE=DYNLINK -c ./src/install.c -o install.2.o  -I./src -I./src/driver -I.
gcc -O2 -Wall -DWINVER=0x500  -Werror -Wno-unknown-pragmas -Wno-multichar -DLOG_APPNAME=\"libusb0-dll\" -DTARGETTYPE=DYNLINK -c ./src/registry.c -o registry.2.o  -I./src -I./src/driver -I.
windres  -I./src ./src/resource.rc -o resource.2.o
gcc -O2 -Wall -DWINVER=0x500  -Werror -Wno-unknown-pragmas -Wno-multichar -DLOG_APPNAME=\"libusb0-dll\" -DTARGETTYPE=DYNLINK -o libusb0.dll -I./src  usb.2.o error.2.o descriptors.2.o windows.2.o install.2.o registry.2.o resource.2.o libusb0.def -s -mdll -Wl,--kill-at -Wl,--out-implib,libusb.a -Wl,--enable-stdcall-fixup -L. -lcfgmgr32 -lsetupapi -lgdi32
gcc -O2 -Wall -DWINVER=0x500  -Werror -Wno-unknown-pragmas -Wno-multichar -DLOG_APPNAME=\"install-filter\" -DTARGETTYPE=PROGRAMconsole -DLOG_STYLE_SHORT -c ./src/install_filter.c -o install_filter.1.o  -I./src -I./src/driver -I.
gcc -O2 -Wall -DWINVER=0x500  -Werror -Wno-unknown-pragmas -Wno-multichar -DLOG_APPNAME=\"install-filter\" -DTARGETTYPE=PROGRAMconsole -DLOG_STYLE_SHORT -c ./src/error.c -o error.1.o  -I./src -I./src/driver -I.
gcc -O2 -Wall -DWINVER=0x500  -Werror -Wno-unknown-pragmas -Wno-multichar -DLOG_APPNAME=\"install-filter\" -DTARGETTYPE=PROGRAMconsole -DLOG_STYLE_SHORT -c ./src/install.c -o install.1.o  -I./src -I./src/driver -I.
gcc -O2 -Wall -DWINVER=0x500  -Werror -Wno-unknown-pragmas -Wno-multichar -DLOG_APPNAME=\"install-filter\" -DTARGETTYPE=PROGRAMconsole -DLOG_STYLE_SHORT -c ./src/registry.c -o registry.1.o  -I./src -I./src/driver -I.
windres  -I./src ./src/install_filter_rc.rc -o install_filter_rc.1.o
gcc -O2 -Wall -DWINVER=0x500  -Werror -Wno-unknown-pragmas -Wno-multichar -DLOG_APPNAME=\"install-filter\" -DTARGETTYPE=PROGRAMconsole -DLOG_STYLE_SHORT -o install-filter.exe -I./src  install_filter.1.o error.1.o install.1.o registry.1.o install_filter_rc.1.o -s -L. -lgdi32 -luser32 -lcfgmgr32 -lsetupapi
gcc -c ./tests/testlibusb.c -o testlibusb.3.o -O2 -Wall -DWINVER=0x500  -Werror -Wno-unknown-pragmas -Wno-multichar -DLOG_APPNAME=\"testlibusb\" -DTARGETTYPE=PROGRAMconsole  -I./src -I./src/driver -I.
./tests/testlibusb.c: In function 'print_device':
./tests/testlibusb.c:95:63: error: ' - ' directive output may be truncated writing 3 bytes into a region of size between 1 and 256 [-Werror=format-truncation=]
   95 |                 snprintf(description, sizeof(description), "%s - ", string);
      |                                                               ^~~
./tests/testlibusb.c:95:17: note: 'snprintf' output between 4 and 259 bytes into a destination of size 256
   95 |                 snprintf(description, sizeof(description), "%s - ", string);
      |                 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
cc1.exe: all warnings being treated as errors
make: *** [Makefile:162: testlibusb.3.o] Error 1

```

mcuee, 2024-02-09:
gcc version.
```
$ gcc -v
Using built-in specs.
COLLECT_GCC=C:\msys64\ucrt64\bin\gcc.exe
COLLECT_LTO_WRAPPER=C:/msys64/ucrt64/bin/../lib/gcc/x86_64-w64-mingw32/13.2.0/lto-wrapper.exe
Target: x86_64-w64-mingw32
Configured with: ../gcc-13.2.0/configure --prefix=/ucrt64 --with-local-prefix=/ucrt64/local --build=x86_64-w64-mingw32 --host=x86_64-w64-mingw32 --target=x86_64-w64-mingw32 --with-native-system-header-dir=/ucrt64/include --libexecdir=/ucrt64/lib --enable-bootstrap --enable-checking=release --with-arch=nocona --with-tune=generic --enable-languages=c,lto,c++,fortran,ada,objc,obj-c++,jit --enable-shared --enable-static --enable-libatomic --enable-threads=posix --enable-graphite --enable-fully-dynamic-string --enable-libstdcxx-filesystem-ts --enable-libstdcxx-time --disable-libstdcxx-pch --enable-lto --enable-libgomp --disable-libssp --disable-multilib --disable-rpath --disable-win32-registry --disable-nls --disable-werror --disable-symvers --with-libiconv --with-system-zlib --with-gmp=/ucrt64 --with-mpfr=/ucrt64 --with-mpc=/ucrt64 --with-isl=/ucrt64 --with-pkgversion='Rev4, Built by MSYS2 project' --with-bugurl=https://github.com/msys2/MINGW-packages/issues --with-gnu-as --with-gnu-ld --disable-libstdcxx-debug --with-boot-ldflags=-static-libstdc++ --with-stage1-ldflags=-static-libstdc++
Thread model: posix
Supported LTO compression algorithms: zlib zstd
gcc version 13.2.0 (Rev4, Built by MSYS2 project)
```

dontech, 2024-02-09:
fixed with: https://github.com/mcuee/libusb-win32/commit/358ef3718952956751d869f52979cd38282d54ee

dontech, 2024-02-09:
Considered closed. Reopen if its not.

dontech, 2024-02-09:
Closed with https://github.com/mcuee/libusb-win32/commit/358ef3718952956751d869f52979cd38282d54ee

closed 2024-02-09 by dontech

mcuee, 2024-02-11:
Yes git is fine now.
```
MINGW64 /c/work/libusb/libusb-win32/libusb-win32/libusb
$ make
gcc -O2 -Wall -DWINVER=0x500  -Werror -Wno-unknown-pragmas -Wno-multichar -DLOG_APPNAME=\"libusb0-dll\" -DTARGETTYPE=DYNLINK -c ./src/usb.c -o usb.2.o  -I./src -I./src/driver -I.
gcc -O2 -Wall -DWINVER=0x500  -Werror -Wno-unknown-pragmas -Wno-multichar -DLOG_APPNAME=\"libusb0-dll\" -DTARGETTYPE=DYNLINK -c ./src/error.c -o error.2.o  -I./src -I./src/driver -I.
gcc -O2 -Wall -DWINVER=0x500  -Werror -Wno-unknown-pragmas -Wno-multichar -DLOG_APPNAME=\"libusb0-dll\" -DTARGETTYPE=DYNLINK -c ./src/descriptors.c -o descriptors.2.o  -I./src -I./src/driver -I.
gcc -O2 -Wall -DWINVER=0x500  -Werror -Wno-unknown-pragmas -Wno-multichar -DLOG_APPNAME=\"libusb0-dll\" -DTARGETTYPE=DYNLINK -c ./src/windows.c -o windows.2.o  -I./src -I./src/driver -I.
gcc -O2 -Wall -DWINVER=0x500  -Werror -Wno-unknown-pragmas -Wno-multichar -DLOG_APPNAME=\"libusb0-dll\" -DTARGETTYPE=DYNLINK -c ./src/install.c -o install.2.o  -I./src -I./src/driver -I.
gcc -O2 -Wall -DWINVER=0x500  -Werror -Wno-unknown-pragmas -Wno-multichar -DLOG_APPNAME=\"libusb0-dll\" -DTARGETTYPE=DYNLINK -c ./src/registry.c -o registry.2.o  -I./src -I./src/driver -I.
windres  -I./src ./src/resource.rc -o resource.2.o
gcc -O2 -Wall -DWINVER=0x500  -Werror -Wno-unknown-pragmas -Wno-multichar -DLOG_APPNAME=\"libusb0-dll\" -DTARGETTYPE=DYNLINK -o libusb0.dll -I./src  usb.2.o error.2.o descriptors.2.o windows.2.o install.2.o registry.2.o resource.2.o libusb0.def -s -mdll -Wl,--kill-at -Wl,--out-implib,libusb.a -Wl,--enable-stdcall-fixup -L. -lcfgmgr32 -lsetupapi -lgdi32
gcc -O2 -Wall -DWINVER=0x500  -Werror -Wno-unknown-pragmas -Wno-multichar -DLOG_APPNAME=\"install-filter\" -DTARGETTYPE=PROGRAMconsole -DLOG_STYLE_SHORT -c ./src/install_filter.c -o install_filter.1.o  -I./src -I./src/driver -I.
gcc -O2 -Wall -DWINVER=0x500  -Werror -Wno-unknown-pragmas -Wno-multichar -DLOG_APPNAME=\"install-filter\" -DTARGETTYPE=PROGRAMconsole -DLOG_STYLE_SHORT -c ./src/error.c -o error.1.o  -I./src -I./src/driver -I.
gcc -O2 -Wall -DWINVER=0x500  -Werror -Wno-unknown-pragmas -Wno-multichar -DLOG_APPNAME=\"install-filter\" -DTARGETTYPE=PROGRAMconsole -DLOG_STYLE_SHORT -c ./src/install.c -o install.1.o  -I./src -I./src/driver -I.
gcc -O2 -Wall -DWINVER=0x500  -Werror -Wno-unknown-pragmas -Wno-multichar -DLOG_APPNAME=\"install-filter\" -DTARGETTYPE=PROGRAMconsole -DLOG_STYLE_SHORT -c ./src/registry.c -o registry.1.o  -I./src -I./src/driver -I.
windres  -I./src ./src/install_filter_rc.rc -o install_filter_rc.1.o
gcc -O2 -Wall -DWINVER=0x500  -Werror -Wno-unknown-pragmas -Wno-multichar -DLOG_APPNAME=\"install-filter\" -DTARGETTYPE=PROGRAMconsole -DLOG_STYLE_SHORT -o install-filter.exe -I./src  install_filter.1.o error.1.o install.1.o registry.1.o install_filter_rc.1.o -s -L. -lgdi32 -luser32 -lcfgmgr32 -lsetupapi
gcc -c ./tests/testlibusb.c -o testlibusb.3.o -O2 -Wall -DWINVER=0x500  -Werror -Wno-unknown-pragmas -Wno-multichar -DLOG_APPNAME=\"testlibusb\" -DTARGETTYPE=PROGRAMconsole  -I./src -I./src/driver -I.
gcc -O2 -Wall -DWINVER=0x500  -Werror -Wno-unknown-pragmas -Wno-multichar -DLOG_APPNAME=\"testlibusb\" -DTARGETTYPE=PROGRAMconsole -o testlibusb.exe -I./src  testlibusb.3.o -s -L. -lusb -lgdi32 -luser32 -lcfgmgr32 -lsetupapi -lcomctl32
gcc -c ./tests/testlibusb_win.c -o testlibusb_win.4.o -O2 -Wall -DWINVER=0x500  -Werror -Wno-unknown-pragmas -Wno-multichar -DLOG_APPNAME=\"testlibusb-win\" -DTARGETTYPE=PROGRAMwindows  -I./src -I./src/driver -I.
windres  -I./src ./tests/testlibusb_win_rc.rc -o testlibusb_win_rc.4.o
gcc -O2 -Wall -DWINVER=0x500  -Werror -Wno-unknown-pragmas -Wno-multichar -DLOG_APPNAME=\"testlibusb-win\" -DTARGETTYPE=PROGRAMwindows -o testlibusb-win.exe -I./src  testlibusb_win.4.o testlibusb_win_rc.4.o -s -L. -lusb -lkernel32 -lgdi32 -luser32 -lnewdev -lsetupapi -lcomctl32 -lole32 -mwindows
```