mcuee/libusb-win32 #63: Using Win8 features requires allocation changes

tormodvolden, 2024-02-11:
As soon as we start using USBD_CLIENT_CONTRACT_VERSION_602 in order to access USBD_HANDLE [1] and the new stuff in usbdex.lib, we also have to fulfill the "contract" [2], and use the Win8 USB allocation routines [3] etc.

[1] https://learn.microsoft.com/en-us/windows-hardware/drivers/ddi/usbdlib/nf-usbdlib-usbd_createhandle
[2] https://learn.microsoft.com/en-us/windows-hardware/drivers/usbcon/usb-client-driver-contract-in-windows-8
[3] https://learn.microsoft.com/en-us/windows-hardware/drivers/usbcon/how-to-add-xrb-support-for-client-drivers

mcuee, 2024-02-11:
Interesting. Last time Travis and I did test libusb0.sys and libusbK.sys (along with WinUSB) under Windows 10/11, using libusbK.dll.

tormodvolden, 2024-02-11:
We only started using the "contract" features in commit 65f9a19, two days ago. And it is not like you will necessarily see any problem, Windows will just not guarantee anything and there might be "bug checks".

dontech, 2024-02-12:
Tormod, again you are spot on. Super annoying :-) :+1: 

However, unless the driver verifyer starts complaining, I am going to try to push these changes until we deprecate Windows 7 support.

For now, it would seem it still works fine to use the old APIs. I guess MS probably made it backward compatible even though the docs suggest that it is not recommended.

All my testing shows that it seems to work.

dontech, 2024-02-12:
Additionally it is worth noticing that this is not a win8 feature, but actually is supported all the way back to Vista.
It was introduced in the win8 SDK, but was O/S backported to win7/vista.
So the binary interface is only missing in older versions of those O/S

The docs state:

`Requires WDK for Windows 8. Targets Windows Vista and later versions of the Windows operating system.`

tormodvolden, 2024-02-12:
Right. When linking in the usbdex.lib, these features will work on Vista/Win7 as well. So when I was saying "build for pre-win8" I should rather have said "build with pre-win8 SDK". Which for now unfortunately affects MinGW. If usbdex was a run-time library like usbd.sys, we could have worked this out for MinGW quite easily. But with the statically linked usbdex.lib it gets more hairy: MinGW doesn't ship any equivalent functions or library. And if linking in an usbdex.lib (I tried, I found one on a machine even without having the SDK on it), it has a lot of new dependencies: