mcuee/libusb-win32 #69: usb_find_devices() return 0

dythj, 2024-04-23:
when I calling interface usb_find_devices(),it return 0. But it works on the other computer and return 1

mcuee, 2024-04-23:
Did you install the libusb0.sys driver for at least one device?

dythj, 2024-04-23:
Do you mean I need to run install-filter.exe to install the driver?

I ran install-filter.exe and install-filter-win.exe,  install-filter.exe was crashed, install-filter-win.exe installed the driver, but my device stopped working

dythj, 2024-04-23:
Here's a comparison of the two results

[image]

[image]

mcuee, 2024-04-23:
filter driver is not guarateed to work. Please post your device details to the following issue.
* https://github.com/mcuee/libusb-win32/issues/35

Are you saying that the filter driver version 1.2.6 works with the USB keyboard but not 1.4.0 version?

Does device driver mode work? You need to use Zadig to install the device driver (replacing the system USB keyboard driver).

dythj, 2024-04-23:
Quote: Are you saying that the filter driver version 1.2.6 works with the USB keyboard but not 1.4.0 version?

These two images are the result of two computers running out, one computer can find the device, the other can't

Quote:Does device driver mode work? You need to use Zadig to install the device driver (replacing the system USB keyboard driver).

I tried to use the Zadig software to reverse the driver, but I plugged the device into the computer and couldn't find the device

[image]

dythj, 2024-04-23:
When I install WinUSB driver on my device, the device will not run, but installed libusb-win32 driver will run, do you know the difference between the two drivers?

mcuee, 2024-04-23:
> When I install WinUSB driver on my device, the device will not run, but installed libusb-win32 driver will run, do you know the difference between the two drivers?

Device driver will replace the original driver -- so your keyboard function will be gone.

Anyway, if you want to play with the USB Keyboard, please do not use libusb-win32, it is not the right library.

dythj, 2024-04-23:
Ok,I will try it.Anyway, problem solved. Thank you！！

closed 2024-04-23 by mcuee