mcuee/libusb-win32 #72: usb_open() failed sometimes

hellomanyi, 2024-08-26:
Hello, I found that there is a low probability that usb_open() cannot be opened when using libusb-win32. After analyzing libusb-1.0 and libusbK, I initially judged that the opening mode of CreateFile may cause it to fail to open; libusb-win32 uses exclusive mode; both libusb-1.0 and libusbK are opened in shared mode; can we modify this code in the official version?

libusb-win32：
CreateFile(dev_name, 0, 0, NULL, OPEN_EXISTING,FILE_FLAG_OVERLAPPED, NULL);

libusb-1.0
  handle = CreateFileA(path, access, FILE_SHARE_READ | FILE_SHARE_WRITE, NULL, OPEN_EXISTING, FILE_FLAG_OVERLAPPED, NULL);

Libusbk
  CreateFileA(DevicePathStrA, GENERIC_READ | GENERIC_WRITE, FILE_SHARE_READ | FILE_SHARE_WRITE, NULL, OPEN_EXISTING, FILE_FLAG_OVERLAPPED, NULL)

mcuee, 2024-09-11:
What is the device in question?

dontech, 2024-12-21:
Hi hellomanyi,

It is not a mistake that only one program can open a device. If multiple programs use the same USB device, the outcome will be fairly random. I do not know why libusb-1.0 and libusbK use SHARED mode, but in most cases it is a really bad idea.

Most USB operations are peer-to-peer. If more than one program controls the device, the behavior can become really funky. Imagine one program switches the "alt-mode" of an interface, without the other program knowing about it while it is reading/writing to the interface. Chaos will occur.

A program using using libusb/libusbk/libusb-1.0 or similar should create a service or similar to relay the traffic to the unit, so there is only one context controlling the device at all times.

If the open fails once in a while, it is probably because you have another device frequently opening the device behind your back. This is fatal.

My advice would be to use debugview to see if someone is actually opening the device.
https://learn.microsoft.com/en-us/sysinternals/downloads/debugview

Then use procmon to figure out who is opening the device:
https://learn.microsoft.com/en-us/sysinternals/downloads/procmon

I am not saying there cannot be a bug in libusb, but it is more likely a case of double usage of the same device.

Thanks,

/pedro

closed 2025-03-21 by mcuee