mcuee/libusb-win32 #73: ntdef.h(197,1): error C1189: #error:  "No Target Architecture"

durongze, 2024-12-21:
1>abort_endpoint.c
16:42:26:068	1>C:\Program Files (x86)\Windows Kits\10\Include\10.0.22621.0\shared\ntdef.h(197,1): error C1189: #error:  "No Target Architecture"
16:42:26:068	1>claim_interface.c
16:42:26:130	1>C:\Program Files (x86)\Windows Kits\10\Include\10.0.22621.0\shared\ntdef.h(197,1): error C1189: #error:  "No Target Architecture"
16:42:26:130	1>clear_feature.c
16:42:26:192	1>C:\Program Files (x86)\Windows Kits\10\Include\10.0.22621.0\shared\ntdef.h(197,1): error C1189: #error:  "No Target Architecture"
16:42:26:192	1>dispatch.c
16:42:26:254	1>C:\Program Files (x86)\Windows Kits\10\Include\10.0.22621.0\shared\ntdef.h(197,1): error C1189: #error:  "No Target Architecture"
16:42:26:254	1>driver_debug.c
16:42:26:254	1>C:\Program Files (x86)\Windows Kits\10\Include\10.0.22621.0\shared\ntdef.h(197,1): error C1189: #error:  "No Target Architecture"
16:42:26:314	1>driver_registry.c
16:42:26:314	1>C:\Program Files (x86)\Windows Kits\10\Include\10.0.22621.0\shared\ntdef.h(197,1): error C1189: #error:  "No Target Architecture"
16:42:26:315	1>get_configuration.c
16:42:26:376	1>C:\Program Files (x86)\Windows Kits\10\Include\10.0.22621.0\shared\ntdef.h(197,1): error C1189: #error:  "No Target Architecture"
16:42:26:376	1>get_descriptor.c
16:42:26:439	1>C:\Program Files (x86)\Windows Kits\10\Include\10.0.22621.0\shared\ntdef.h(197,1): error C1189: #error:  "No Target Architecture"
16:42:26:439	1>get_interface.c
16:42:26:515	1>C:\Program Files (x86)\Windows Kits\10\Include\10.0.22621.0\shared\ntdef.h(197,1): error C1189: #error:  "No Target Architecture"

durongze, 2024-12-21:
[image]
Microsoft Visual Studio Community 2022
Version 17.10.5
VisualStudio.17.Release/17.10.5+35122.118
Microsoft .NET Framework
Version 4.8.09037

Installed Version: Community

Visual C++ 2022   00482-90000-00000-AA053
Microsoft Visual C++ 2022

ASP.NET and Web Tools   17.10.344.38934
ASP.NET and Web Tools

Azure App Service Tools v3.0.0   17.10.344.38934
Azure App Service Tools v3.0.0

C# Tools   4.10.0-3.24324.8+04fb39164c99c519832109f21b22630bc9246fce
C# components used in the IDE. Depending on your project type and settings, a different version of the compiler may be used.

Debugging Tools for Windows   10.0.26100.1
Integrates the Windows Debugger functionality (http://go.microsoft.com/fwlink/?linkid=223405) in Visual Studio.

GitHub Copilot   0.2.894.62884
GitHub Copilot is an AI pair programmer that helps you write code faster and with less work.

Microsoft JVM Debugger   1.0
Provides support for connecting the Visual Studio debugger to JDWP compatible Java Virtual Machines

NuGet Package Manager   6.10.2
NuGet Package Manager in Visual Studio. For more information about NuGet, visit https://docs.nuget.org/

Test Adapter for Boost.Test   1.0
Enables Visual Studio's testing tools with unit tests written for Boost.Test.  The use terms and Third Party Notices are available in the extension installation directory.

Test Adapter for Google Test   1.0
Enables Visual Studio's testing tools with unit tests written for Google Test.  The use terms and Third Party Notices are available in the extension installation directory.

TypeScript Tools   17.0.30327.2001
TypeScript Tools for Microsoft Visual Studio

Visual Basic Tools   4.10.0-3.24324.8+04fb39164c99c519832109f21b22630bc9246fce
Visual Basic components used in the IDE. Depending on your project type and settings, a different version of the compiler may be used.

Visual F# Tools   17.10.0-beta.24228.1+dd749058c91585e9b5dae62b0f8df892429ee28f
Microsoft Visual F# Tools

Visual Studio IntelliCode   2.2
AI-assisted development for Visual Studio.

Windows Driver Kit   10.0.26100.1
Headers, libraries, and tools needed to develop, debug, and test Windows drivers (msdn.microsoft.com/en-us/windows/hardware/gg487428.aspx)

durongze, 2024-12-21:
x86：_X86_

dontech, 2024-12-21:
Hi durongze,

1) Official builds are only supported via the cmd line interface, which build using the EWDK.
2) Debugging is only supported via vs2019.

Both of these should run more or less out-of-the-box.

What are you trying to debug?

Thanks,

/pedro

dontech, 2024-12-21:
As soon as we drop support for Windows-7 we can switch to newer versions of VS/EWDK. But for now we stick with vs2019.

Thanks,

/pedro

durongze, 2024-12-23:
感谢回复

closed 2024-12-23 by durongze

dontech, 2024-12-29:
没问题

No problem, any time. :-)

/pedro