mcuee/libusb-win32 #74: Can't install libusb

Adaaks, 2025-01-05:
How do you install this? I have downloaded the program install-filter-win.exe, but when I find my printer and try install libusb, then my printer automatically goes offline when I install libusb, only when I remove it then my printer comes back. This means I can't discover the printer in brother_ql
  File "C:\Users\adamp\AppData\Local\Programs\Python\Python310\lib\site-packages\brother_ql\backends\__init__.py", line 38, in backend_factory
    raise NotImplementedError('Backend %s not implemented.' % backend_name)
NotImplementedError: Backend None not implemented.

mcuee, 2025-01-07:
Filter driver is not guaranteed to work. If it does not work, you have to try device driver mode.

mcuee, 2025-01-07:
The code in https://github.com/pklaus/brother_ql seems to be very old as well. You may have to get the support from the project itself.