mcuee/libusb-win32 #77: USB 2.0 : receiving 512 then ZLP blocks for ever when trying to read N x 512 (with N>1)

sonatique, 2025-08-15:
Hello @dontech ,

I am facing an issue that seems surreal, using latest libusb , libusbK and libusb0.sys . 

By carefully adding logs in all these 3 places, and simplifying my scenario, I am now pretty convinced that the issue is in libusb0.sys.

In short: when trying to receive 512 bytes + ZLP in a buffer of 1024, the callback transfer is never called and everything freezes. Everything works for 512 bytes + 4 bytes.

Here is my scenario:

I am first submitting a read transfer for 1024 bytes. libusb0.sys (compiled in Debug) outputs this is DebugView:
`libusb0-sys:[transfer] [bulk-read #206] EP86h length=1024, packetSize=512, maxTransferSize=4194304`
-> OK.

Then my device do something (I double-checked this using a USB sniffer): it first sends a 512 bytes IN transfer, then a IN ZLP to signal that there are no more data.
 
I see something regarding reception that first looks good:
`libusb0-sys:[transfer_complete] sequence 206: 512 bytes transmitted, maximum_packet_size=512, totalLength=1024`

But then: nothing. The completion routine in libusb0.sys is never called again and the libusb transfer callback is never called. I am expecting the ZLP to signal the end and the user callback to be called. And really: I am 100% sure the ZLP went over the wire.

I had a look a the code. I am seeing:
```
if(NT_SUCCESS(irp->IoStatus.Status)
		&& USBD_SUCCESS(c->urb->UrbHeader.Status)
		&& (c->urb->UrbHeader.Function == URB_FUNCTION_BULK_OR_INTERRUPT_TRANSFER)
		&& !(transmitted % c->maximum_packet_size)
		&& c->totalLength)
```
in transfer_complete implementation. The body of the "if" is basically to resubmit in Irp for the expected remainder (512 bytes in this case). Transmitted == 512 in my case, maximum_packet_size == 512 and totalLength == 512 as well because the value has been update a few lines above  with 1024 - 512.

From what I understand, the condition is written such that the Irp resubmission will occur when totalLength (i.e. the number of bytes still to be received ideally) is > 0 and when the received byte count (transmitted) is a multiple of the max packet size.
"transmitted" can not be > max packet size obviously. So it can be:
> 0 and < max packet size : this was a short packet reception, no need to resubmit since no more data
== max packet size:  we shall resubmit in case there are more data
== 0 : this is maybe a problem because the condition will be true as well since (0 % 512) == 0, as if "transmitted" was a multiple of max packet size.
This means that the code will wrongly resubmit a transfer for >0 bytes, when the bus just told us there are no more data.

At first I thought this was the reason for the issue: first, transfer_complete would do the job correctly, then would be called again with transmitted == 0, would submit another Irp and would wait for something to happen in vain (since the device doesn't send anything anymore).

BUT: I realized that transfer_complete is not called a second time. And therefore I think you knew transfer_complete would never be called for transmitted == 0, such that you knew that writing ` !(transmitted % c->maximum_packet_size)` would be a correct not-short packet detection condition given transmitted is constrained to >0 and <= 512. Note that I would suggest something something simply like transmitted == maximum_packet_size for detecting not-short packet, anyway.

But then I am puzzled:  if transfer_complete is never called for ZLP: how could the code unwind when it receives a ZLP for signaling the end then calls the user callback with just 512 bytes of received data?

I think there is an issue, but I so far I could not figure it out. 
Does the code just not support getting a number of bytes equal to maximum_packet_size then a ZLP?

Note that everything works as expected when my device sends 512 bytes then 4 bytes for instance: transfer_complete is called a second time for the 4 bytes: the condition detect the short packet, avoid resubmitting an Irp and everything terminate as expected.

Could you help me figuring out what is happening? Thanks a lot!

sonatique, 2025-08-15:
@dontech : I think that since one single transfer_complete can receive more than max packet size bytes at once, there is probably no reliable way to understand whether to resubmit or not. I think there shall be no resubmission unless we received exactly  maxTransferSize bytes and totalLength > maxTransferSize. 

I need to think a little bit more about this, then I'll probably propose a PR.

dontech, 2025-08-21:
OK i have reproduced the problem.

Here is the breakdown:

1) There is no way to detect ZLP in usbd.sys API.

2) The only way to "detect" that a transfer could be the result of ZLP is to extend the check to see if transfer is smaller than maxTransferSize, as this is the official way the driver below signals ZLP.

3) Only catch here is that if transfer is cut short because of anything else, the transfer is stopped here. That is not fatal, as the application above is told the transferred size anyway.

I will submit a change for this, as the current behavior is unacceptable.

dontech, 2025-08-21:
maxTransferSize is calculated according to these rules: https://learn.microsoft.com/en-us/windows-hardware/drivers/usbcon/usb-bandwidth-allocation

dontech, 2025-08-21:
ZLP fixed with: https://github.com/mcuee/libusb-win32/commit/aa869d72dc60171a802c36c34458583ee9c752cc

dontech, 2025-08-21:
The other problem with  8192 "block size" is not reproducible here.

Also according to your post: 

libusb0-sys:[transfer] [bulk-read #206] EP86h length=1024, packetSize=512, maxTransferSize=4194304

maxTransferSize = 4MB, which is much higher than 8192. Are you sure the 8192 limit is not in your device?

If not could you send Debug view logs and USB trace logs?

Thanks,

/pedro

dontech, 2025-08-22:
Hi S,

> I am still wondering why this bug have stayed unnoticed for so long, I am puzzled because once you see it, it seems quite serious.

Hehe, i guess this follows what i call the "lifecycle of a bug" as i call it. First it's "why does this not work", then its "oh, this does not work", and then at last "how did this ever work?".....

I tested the previous release vigorously, and use this project at 100s of clients but still did not notice this. Amazing..... :-) 

Thanks,

/pedro

dontech, 2025-08-22:
I will spin a new release this weekend.

sonatique, 2025-08-22:
> Hehe, i guess this follows what i call the "lifecycle of a bug" as i call it. First it's "why does this not work", then its "oh, this does not work", and then at last "how did this ever work?".....

Yes ;-) also known as "The Six Stages of Debugging": https://mwcremer.blogspot.com/2007/06/six-stages-of-debugging.html?m=1 ;-)
 
> The optimization in the driver that it can chain read/writes on a kernel level is also a bit extreme, but gives a bit extra in some cases.

I had a second thought and I am no longer sure that these are just pretty optimizations. I think with this you implemented the very concept of USB transfer, which is quite important and this is how the host program is expecting things to work, as far as I know.

Thank you for your reactivity and efficiency!

sonatique, 2025-08-22:
> I will spin a new release this weekend.

Cool! Quick question, I saw https://github.com/mcuee/libusb-win32/issues/78 : are you going to release 1.4.0.1 with your "old" toolchain such that the binary will support the same OS and CPU as before?

dontech, 2025-08-26:
This is pending, as i do not have access to my signing token right now. Soon!

Let me know if you need an unsigned build.

dontech, 2025-08-27:
OK, will create this for you, and then i will re-post the release signed when i get back to my signing token.

dontech, 2025-08-28:
Here ya go:

https://github.com/mcuee/libusb-win32/releases/tag/snapshot_1.4.0.1

closed 2026-01-09 by dontech